iliasam/OpenTOFLidar #1: Designator of some components are missing in Gerber Files.

AbidHussain70, 2020-03-23:
Hi, First wanna say thanks. Very well explained and well documented.
I want to make one of these, but when I upload the gerber files it seems some of the components footprints have no designator.
 Below are the pictures of both versions. 
[image]

[image]

[image]
[image]

iliasam, 2020-03-23:
You are right. 
"PCB_project_v1" have less designators is Silkscreen.
"PCB_project_v4" have more designators is Silkscreen. 
Also, there are special images at "PCB_project_v4" folder: "PCB_3D_view_down", "PCB_3D_view_up" with manually written designators.

AbidHussain70, 2020-03-23:
Thanks for quick response.
Yes I saw those images. I want to send the PCB to manufacturer for soldering, anyhow i will try to solder it. 
I have another question, both the version v1 and v4 work fine? 
I want to use v4.

iliasam, 2020-03-23:
See board differences description: https://github.com/iliasam/OpenTOFLidar/blob/develop/PCB/README.md
I'm not sure that Silkscreen marking is enough for PCB manufacturer to assemble the board.
The could ask special assembling drawing or some kind of "pick and place" file.

AbidHussain70, 2020-03-24:
I can get my PCB with these files, but they will not solder it. Because of designator issue. I will try it by myself and I will update you about the results. 
Thanks

AbidHussain70, 2020-03-24:
Did you use QUARTZ_32K in "PCB_project_v4"? I saw the two images "PCB_3D_view_down" and PCB_3D_view_up" and also I check the Gerber files, but I could not find the location of the component in the PCB.

iliasam, 2020-03-24:
Be careful with WIFI in "PCB_project_v4 ". I had to install a special metallic plate between WIFI antenna and LIDAR to reduce noise from WIFI. It is not shown at photos.

closed 2020-03-25 by iliasam