iliasam/OpenTOFLidar #3: Compilation Error on IAR 8.40.1

AbidHussain70, 2020-04-29:
I am encoder disc with 100 slits in it instead of 65 slits. I could not find encoder disc with 65 slits. 
So I changed the parameter "ENCODER_HOLES_CNT" to 100 (reference is attached below)  and tried recompiling the code in the latest IAR IDE but it gives the following error during compilation.  

https://github.com/iliasam/OpenTOFLidar/blob/d2ed102225cc642949f4d09c30fa0e5a3a8fc32a/Firmware/scanning_fw/config.h#L185

**Here is the screen-shoot of the error**

[image]

**When I open your project files**, the IAR software ports the old project file to a updated project file in the new version. Is the error because of this? If so what version you used for the successfully compilation?

iliasam, 2020-04-29:
Yes, this problem is caused by IAR version.
I'm using IAR 7.50.

Try to add this code after "#include "stm32f30x.h"" in main.h:
#if defined ( __ICCARM__ ) // IAR
  #include "intrinsics.h"  
#endif

AbidHussain70, 2020-04-29:
I tired this but I was getting the same error. 

**Then I remove the if statement and it works.**

#include "stm32f30x.h"" in main.h:
//#if defined ( ICCARM ) // IAR
#include "intrinsics.h"
//#endif

Thanks

iliasam, 2020-04-29:
I'm sorry, there was a problem in Github message parser.
The code should be:
`#if defined(__ICCARM__) // IAR`
` #include "intrinsics.h"`
`#endif`
Does it work?

AbidHussain70, 2020-04-29:
Ohhh, no need to sorry. 

First I tried the given command
**#if defined(__ICCARM__) // IAR
#include "intrinsics.h"
#endif**

But it was giving the same error. Then I removed the if statement and it works.
**#include "intrinsics.h"**

iliasam, 2020-04-29:
The will no be "intrinsics.h" in Keil.
I want to change me code to bake it better.
The problem was that is true "`__ICCARM__`" not "ICCARM"
You should see the same definitions below in main.h.

AbidHussain70, 2020-04-29:
ok, I will check it tomorrow again and I will update you.
But with the above mentioned way I got the .hex files.
[image]

closed 2020-05-02 by iliasam

AbidHussain70, 2020-05-08:
> #if defined(__ICCARM__) // IAR
> #include "intrinsics.h"
> #endif

I tried these lines also, but I was getting the same error. 
So I just add the following line,

> #include "intrinsics.h".
And it works.