iliasam/OpenTOFLidar #4: What is the ref_plate_distance

AbidHussain70, 2020-04-29:
Hi,

**Whats this REF_PLATE_DIST variable?** 
https://github.com/iliasam/OpenTOFLidar/blob/d2ed102225cc642949f4d09c30fa0e5a3a8fc32a/Firmware/scanning_fw/config.h#L26

**is it this back plate?**
[image]

**Or is it this top plate distance?**
[image]

iliasam, 2020-04-29:
"Reference plate" is a white plastic plate (you marked it at the first picture). LIDAR could "see" this mate when mirror is turned to zero angle. (At upper pictures mirror is at ~180 deg position).
LIDAR is measuring all distance from the rotation axis in horizontal plane.
So "REF_PLATE_DIST" is distance in mm between the rotation axis and the "Reference plate"

AbidHussain70, 2020-04-29:
Great, I got it.

Thanks

closed 2020-04-29 by AbidHussain70