iliasam/OpenTOFLidar #5: PC Utility test run for Lidar testing and lidar scanning

AbidHussain70, 2020-05-08:
I did my first run but I am not getting the distance measurement. Any clue?
[image]

and I also changed encodre count to 20,
My encoder gives digital output 3v.

[image]

[image]

iliasam, 2020-05-08:
Have you tried to follow  this instructions: https://github.com/iliasam/OpenTOFLidar/wiki/Electronics-assembling-notes ?
Firstly, you should test your device in not-scanning mode: https://github.com/iliasam/OpenTOFLidar/tree/develop/Firmware
Do not try to run scanning mode until you get good results in not-scanning mode.
  
I can see from the first image that the is no received signal or it is too low.
Also I can see that you set APD target voltage to 132V, but measured voltage is 126V.
It would be good to measure breakdown voltage of the APD.
For AD500-8 it is close to 90V, for MTAPD-07-13 it is close to 130V.
But for testing you could set less voltage - to avoid breakdown .

AbidHussain70, 2020-05-09:
Great, it worked.

[image]

[image]

iliasam, 2020-05-09:
I can see from the histogram that received signal is still very small. See  this page with examples: https://github.com/iliasam/OpenTOFLidar/wiki/Resolution-of-the-LIDAR
Do you have an oscilloscope to  see received signal form?

iliasam, 2020-05-13:
Sorry, previous link is for scanning mode.
You should see screenshots of the "Testing Utility".
You can find them here: https://github.com/iliasam/OpenTOFLidar/wiki/LidarTestingUtility
Here: https://ibb.co/vjFfwVD
And in the video: https://youtu.be/lTPH_Xa9yCk

iliasam, 2020-05-17:
Should I close this issue?

AbidHussain70, 2020-05-18:
Hi,
I will do the oscilloscope test and I will post the results.

AbidHussain70, 2020-05-21:
I checked with oscilloscope, with changing Vlaser, the Vpp on connector J2 changes. The width of the pulse also increases with increase in Vlaser. 

> Vlaser 16V, output is below.

.
[image]

> Vlaser 15V oscilloscope result is given below.
[image]

AbidHussain70, 2020-05-21:
The signal strength increases on white surface. But on Random surfaces the strength remains the same. By same I mean the same as shown in the pictures I posted earlier.

iliasam, 2020-05-21:
If you oscilloscope configuration is right, you pictures mean that laser current is too low.
At the first picture one cell is 50mV, so signal peak amplitude is ~150-200mV. Current sensing resistor must be 0.1 ohm, so current is ~1.5-2A. This is too low.
See this picture for a reference: https://github.com/iliasam/OpenTOFLidar/blob/develop/Images/Oscilloscope/LaserCurrent.png
Could you say your oscilloscope model?

AbidHussain70, 2020-05-22:
Sorry I forgot to mention, my oscilloscope probe is 10x. 150~200mV means 1.5~2V.

iliasam, 2020-05-22:
Ok, but what about strong additional noise?
Could you measure TIA signal at J1 ? 
It would be good to measure signals at J1 and J2 simultaneously.

AbidHussain70, 2020-05-26:
I order the connectors, I will test it once I received the connectors.

AbidHussain70, 2020-06-05:
I checked but I am not getting any pulse at TIA J1.

iliasam, 2020-06-05:
Have you tried to check if laser is really working with a phone camera?
It is strange that you can't see any pulses, while you can get some measurements (as you showed earlier).
Amplitude of the pulses should be low - see here: https://ibb.co/m9Yj5TD
It is useful to use J2 laser signal as a trigger for the oscilloscope.

AbidHussain70, 2020-06-09:
[image]

This is laser check with camera. But I noticed it is blinking in irregular interval.

iliasam, 2020-06-09:
> But I noticed it is blinking in irregular interval.
I think that this is OK. It could be produced by camera frame rate.

Could you confirm that you can see that "START" values are changing somehow when you are changing distance to the object (as you said earlier)?
Also, I advice you to make non-scanning tests without mirror (to avoid possible wrong reflections).

AbidHussain70, 2020-06-10:
[image]

I am getting the distance in meters when I move an object away from the scanner. But the measurement is not stable. And also accuracy is not good.

AbidHussain70, 2020-06-10:
I will assemble another scanner.

iliasam, 2020-06-10:
I can see that your photodiode is really receiving light signal and TIA is working somehow.
So it very strange that you can't see TIA signal at the oscilloscope.
Tip: you must use AC coupling it the oscilloscope.
There are two reasons that your received signal is low:
1. Laser is not good aligned. Try to move laser holder against PCB.
2. TIA is particularly damaged. I have seen such case in my practice.

AbidHussain70, 2020-06-10:
[image]
This is what I got when both J1 and J2 is connected with oscilloscope.

iliasam, 2020-06-10:
Is oscilloscope CH2 divider correct?
TIA have ~100mV overdrive limit, so maximum CH2 should be ~100mV.

AbidHussain70, 2020-06-10:
[image]

[image]

**I am not using probe I am using BNC connectors.** 
I got the pulse. Problem was with the UFL connectors.

iliasam, 2020-06-10:
Now it is looking good.
My tests are looking so: https://ibb.co/m9Yj5TD
You can see that at strong overdrive signal length should be > 75 ns

AbidHussain70, 2020-06-10:
[image]

AbidHussain70, 2020-06-10:
How should I update the calibration file? 

I think low values in bin graph is because of calibration.

iliasam, 2020-06-11:
I added addition information to the Wiki about testing PCB without optics, it could be useful fro you: https://github.com/iliasam/OpenTOFLidar/wiki/No-optics-tests

iliasam, 2020-06-16:
Do you have any progress with this problem?

AbidHussain70, 2020-06-17:
Hi, 
I will post an update soon.

AbidHussain70, 2020-06-22:
I did no optics test. 
**Actual distance is 1.2 meter.**  **white wall.** 
[image]
**Laser voltage is 16V.**

[image]
**laser voltage is 15v Actual distance is 1.2 meter. white wall.**

[image]

**laser voltage is 14v Actual distance is 1.2 meter. white wall**

[image]

**laser voltage is 13v Actual distance is 1.2 meter. white wall**

[image]

**laser voltage is 12v Actual distance is 1.2 meter. white wall**

[image]

**laser voltage is 14v Actual distance is 1.2 meter. white wall**

iliasam, 2020-06-22:
It's an amplitude of TIA pulse. It's changing with no optics like this: https://ibb.co/m9Yj5TD

AbidHussain70, 2020-06-22:
but with optics signal amplitude changes when I change the APD voltage.

iliasam, 2020-06-22:
"In my case just pulse width is changing not amplitude."
Maybe something reflecting light from the laser?
When I say - "no optics", I mean that lens holders are installed (without lens).

AbidHussain70, 2020-06-23:
I just remove the optics part not the holder.

iliasam, 2020-06-23:
Maybe APD voltage is too high?

iliasam, 2020-08-03:
How are your results?

Anesthesia-RW, 2020-08-19:
Hi,the non-scan is working but In the scan mode, I get the following result. No point in my result and the state  mask is  0x40. If you give some suggestions I will very appreciation.
[image]

Anesthesia-RW, 2020-08-19:
when i set the APD volt ,there are no change that means the "set" command don't work?
I check the src ,the state mask is 0x40 means there have no CALIBRATION . 
so in scanning model the callbration is what?

iliasam, 2020-08-19:
See info about calibration at this page: https://github.com/iliasam/OpenTOFLidar/wiki/LidarTestingUtility
You need to record data with constant distance, but changing amplitude.

"when i set the APD volt ,there are no change that means the "set" command don't work?"
I'm sorry, I need more information about that you are doing.

Anesthesia-RW, 2020-08-19:
Okay, I will try a few more times, and then record more details of the problem.

Anesthesia-RW, 2020-08-26:
Hello.
Following the steps you gave, I selected the constant distance is 3m and change received signal by slowly closing APD lens with something non-transparentcollect . 
The data and curves are as follows:
[image]
[image]
I sent the  "coff." to MCU, then change the distance to test.
Some data are as follows:
Real_Distance_1--- 5.15m     Test_Distance_1--- 5.95m
Real_Distance_2--- 4.55m     Test_Distance_2--- 5.04m
Real_Distance_3---2.35m      Test_Distance_3---1.95m
Real_Distance_4---1.75m      Test_Distance_4---1.01m
Real_Distance_5---1.2m        Test_Distance_5---0.16m
I found that when Real_Distance<1.1m the Test_Distance is 0.
There a problem with the calibration data. I repeated the calibration more than ten times. Sending  the calibration coefficient  to MCU and test ,the measurement result was quite different.
I use oscilloscope tested J1 and J2,The result is as follows:
--Distance:3m
[image]
[image]

iliasam, 2020-08-26:
Your results are looking strange.
Firstly - I can see that your APD voltage is 63V while target voltage is 128V. This is not normal.  What is the type of your APD?
Are you using the same optics as I am using in my Lidar?

Your calibration data are looking OK, so you can move "Fit Start" line to the 300-350 position before starting "Fit".

Also, after calibration you need to set: " "Ref distance" - distance to a reference object. Used with non-scanning firmware only. You need to enter current distance to the object and press "SET" button". Best distance is 2-3 m.
See https://github.com/iliasam/OpenTOFLidar/wiki/LidarTestingUtility

Anesthesia-RW, 2020-08-27:
Because I can’t find the same APD as yours, no supply. So I changed the type of APD. The optics is the same as yours.

 I have two type APD.  One is MTAPD-07-14 ,the other one is GD5210Y-2 , all of them the APD voltage in range 63~65 (v).
Their performance is very similar to yours MTAPD-07-13

Especially MTAPD-07-14,I checked the information and found that it came from the same data sheet with  MTAPD-07-13.
The GD5210Y-2's data sheet as follows:
[996076217905nm系列APD-规格书.pdf](https://github.com/iliasam/OpenTOFLidar/files/5134165/996076217905nm.APD-.pdf)

iliasam, 2020-08-27:
According to MTAPD-07-14 datasheet its breakdown voltage is in "160-200V" range.
You can see that APD is in breakdown state by seeing a very big noise at the TIA output.
  
According to GD5210Y-2 datasheet its breakdown voltage is in 130-220 V range.
  
In any case, it is **not OK** that measured APD voltage (63V) is not the same as set target voltage (128V). It is look like APD step-up converter or APD voltage measurement circuit are working wrong. 

You should **ensure** that the is no light is getting directly from laser to the APD (or APD lens) - by some holes in the lens holders or by scattering from mirror or mechanics.
I advice you to do calibration and all not scanning tests **without mirror** - lenses should be directly pointed to the target object. I mean this setup: https://habrastorage.org/getpro/habr/post_images/120/c20/e43/120c20e43dc0d41ada95b6ed9ecd82bf.jpg

Anesthesia-RW, 2020-08-27:
I sorry, I forgot to tell you that the earlier calibration are under without mirror conditions .
I mean all test as you advise, constant distance is 3m and without mirror.

[image]

Anesthesia-RW, 2020-08-27:
Could it be a problem with my electronic components?
But all I follow your BOM table, maybe I need to check again and make a new PCB board.
I have made three PCB boards, one of the TIA is damaged. the voltage of the other two boards is about 63v.

iliasam, 2020-08-27:
You cant try to switch to the manual PWM controlling by setting "APD PWM Steps" - second "SET" buttom at the "Rangefinder State" group. Try to change this value starting from low (like 30 steps) to bigger value.

Anesthesia-RW, 2020-08-27:
Change the ”APD PWM“ and recording "APD Volt", the curve as follow:
When the APD PWM >85 , Changing the APD PWM  and the APD volt  changes very little. Fluctuate around 64
[image]

iliasam, 2020-08-27:
This is very strange. Have you tried to measure real APD voltage by voltmeter?

Anesthesia-RW, 2020-08-28:
I can't remove R1. 
Using voltmeter to measure the real APD voltage. I found the voltmeter's vlaue less than soft_APD voltage(MCU get)

iliasam, 2020-08-28:
"I can't remove R1."
Why?
" I found the voltmeter's vlaue less than soft_APD voltage"
How big is the difference?

Anesthesia-RW, 2020-08-28:
[image]

Anesthesia-RW, 2020-08-28:
The difference is in range (1,2).

Anesthesia-RW, 2020-08-28:
Just now.I removed R1 and test again no change in voltage.

iliasam, 2020-08-28:
"The difference is in range (1,2)."
Do you mean volts? If it is right, is is bad - voltmeter have own resistance, an power  of the APD step-up converter is very low, so voltmeter create some voltage drop. Also, APD_voltage is not measured very accurately. 

I can't suggest why APD voltage did not grow bigger that 70V. It look like some HW problem in APD voltage source.

Anesthesia-RW, 2020-08-28:
Yes,the difference is  1v to 2 v.
Thanks for your reply.
I decided to remake a few PCB boards.
How to explain that the previous two pieces have the same error? 
Maybe problems with electronic components?

iliasam, 2020-08-28:
"Maybe problems with electronic components?"
Yes, looks like that some of the components is damaged or wrong.
Are you sure that max voltage parameter of the capacitors C10/C11/C33  is 250V ? 
Is L5 inductance really 47uH ?

I don't think that there are problems with PCB itself.

Anesthesia-RW, 2020-08-28:
I am also troubled by this problem. The electronic components is provided by a specialized manufacturer.

iliasam, 2020-08-28:
But are you sure that **your** BOM (that you send to the components seller) is correct?

Anesthesia-RW, 2020-08-29:
Hi，The APD power is generate by this  part  circuit?
[image]

iliasam, 2020-08-29:
Yes.
R8 and R9 are used for measurement voltage by MCU.

[38 more replies]

closed 2020-10-04 by iliasam