iliasam/OpenTOFLidar #8: The pulse on LASER_PULSE net should be like this？

br76530067, 2021-08-06:
I soldered these components on PCB under your instruction note " #2. install TDC, its quartz. Enable "Auto Request Batch" in utility. There must be pulses at "LASER_PULSE" net. ", and I use an oscilloscope to see the signal on LASER_PULSE net, are these data in the following picture correct?
The test PC utility:
[image]

The oscilloscope:
[image]

Thank you in advance!

iliasam, 2021-08-06:
Pulses shoud have ~1 ms period and they should have 3.3V amplitude.
I can see from your screenshot that "state mask" is 0x5. Description is here: https://github.com/iliasam/OpenTOFLidar/wiki/State-Mask-bits-description
You have bit 0 set which mean that MCU can't detect TDC at startup.
Have you installed all componencs aroung TDC?

br76530067, 2021-08-07:
Hi, iliasam, thank you for your reply
I installed all components around TDC, and I checked the voltage level of every pin of that, it looks like normal.

#1  And I found the temperature of the soldering for TDC is limited to 260 °C in the datasheet, due to manual soldering, I used 350°C to solder TDC, which may cause the broken of TDC?  have you ever met this situation when you solder TDC?

#2 Or do you have some way to check if TDC is broken?

br76530067, 2021-08-07:
Hi ilisam 
it looks like get some progree, however, i cant catch the pulse on LASER_PULSE net.
[image]
[image]

iliasam, 2021-08-07:
Pulses should be short, with 1us width. Set trigger to 1V at your oscilloscope.
I don't know how to find out that TDC is broken.

br76530067, 2021-08-07:
Hi, iliasam,

I under your hint, and get this picture in the oscilloscope, but it looks like only could get one time of rising edge when power is on. Then can not get any pulse anymore, is it correct?
[image]

Thanks in advance

iliasam, 2021-08-07:
Pulses should be present continuously when "Auto Request Batch" flag is set.

br76530067, 2021-08-08:
After changing a TDC-GP21, it works, Thanks, I will continue.

closed 2021-08-08 by br76530067