iliasam/OpenTOFLidar #10: Step 6.Install APD DC-DC co......

br76530067, 2021-08-13:
When doing "6.Install APD DC-DC components (Q1 and others). Check "APD_V" - it must change by manual commands from utility."

I found I only could adjust the APD Volt in scope 0 ~ 59.8V, like below pictures, whatever I changed the APD PWM and APD Targ.Volt. Should it be correct?
[image]

[image]

[image]

[image]

[image]

[image]

iliasam, 2021-08-13:
Have you measured APD voltage by voltmeter. Is the value the same as displayed in utility?
Also you can check this (type of inductor): https://github.com/iliasam/OpenTOFLidar/issues/5#issuecomment-690827160

br76530067, 2021-08-13:
Yes, it is same although some difference, for example, 12.5V in utility, 13.6V in voltmeter.
Let me check #5

br76530067, 2021-08-13:
"In any case, it is not OK that measured APD voltage (63V) is not the same as set target voltage (128V). It is look like APD step-up converter or APD voltage measurement circuit are working wrong." as your repled in #5, let me check if it is .

iliasam, 2021-08-13:
13.6 / 12.5V = 1.088, -> ~9% error, which is big. Can you measure difference at higher voltage  (40-50V)?

br76530067, 2021-08-13:
when  47.3V in utility, 46.4V in voltmeter

br76530067, 2021-08-13:
"Don't try to switch to Auto mode until you get well working manual mode." you mean first use the PWM duty to get higer voltage then use the AUTO mode to set?

br76530067, 2021-08-13:
And one more : when 13.8V in utility, 11.9V in voltmeter

br76530067, 2021-08-13:
"Hello.The APD voltage's problem solved. The problem is the inductor L5 type not value.
The manufacturer gave a multilayer inductor, but the actual winding type. There is no description about L5 type in the BOM."
I think it is same issue, I will buy this one and to try.

iliasam, 2021-08-13:
"you mean first use the PWM duty to get higer voltage then use the AUTO mode to set?"
You need to get stable high voltage  (< 110V) by setting PWM duty manually.

br76530067, 2021-08-17:
Hi, iliasam,
Thanks for your advice, after changing to the leading inductor, I got below picture:
[image]
[image]

[image]

[image]

[image]

And after changing to the high precise resister, I got the 103.7V on voltmeter and 103.1V on PC utility.

br76530067, 2021-08-17:
but what max volt I could achieve is 122V, is it correct?

iliasam, 2021-08-17:
Breakdown voltage of MTAPD-07-013 is in 120-130V.
When APD is installed, you can the that APD is close to breakdown by seeing more noise at the TIA output.
Also, APD voltage would increase slowly (comparing to low voltage - not breakdown mode) as shown here: https://github.com/iliasam/OpenTOFLidar/blob/develop/Images/APD_voltage_vs_PWM.png (for AD500-8).

iliasam, 2021-08-17:
But it is strange that you can't get 130V. Do you have APD installed or not?

br76530067, 2021-08-17:
No, I havent installed the APD, and my APD tupe is MTAPD-07-007.

What is the maximum PWM value that you have set?
200 steps, as above picture.

br76530067, 2021-08-17:
[image]

br76530067, 2021-08-17:
MTAPD-07-013 is on the delivery way

iliasam, 2021-08-17:
The plot that I mentioned earlier, was measured with AD500-8 which have breakdown voltage ~95V. This plot is for information only.
This image is from "Issue 5": https://user-images.githubusercontent.com/48154419/92838938-f75e5f80-f411-11ea-96e4-dbe952ec430b.png
You can see that "Anesthesia-RW" can get 124.5V at 72 PWM steps.
I get 130V at 73V: https://ibb.co/vjFfwVD
All these measurements were made with APD installed, but I don't think that it is important.

Cleaning PCB could be important. 
Check R8, R9 values, check that C10, C11 are high-voltage capacitors.

br76530067, 2021-08-17:
After clearing the PCB, I could get the maximum volt is 126V.
I just put here my soldered components, maybe you have some clue.
[image]

iliasam, 2021-08-17:
Your Q1 and D1 are looking OK.
You can try to shorten L4, but I don't think that this will help.
Maybe it is really problem with C10,C11 working woltage?

br76530067, 2021-08-18:
Because in china market, the  MTAPD-07-013 needs too long to get, and there is no AD500-8, So I buy a AD500-9 to try.
[image]

Do you think the AD500-9 will work?

iliasam, 2021-08-18:
Why there is no AD500-8? https://aliexpress.com/item/32396227457.html
Also I can see them in Taobao.
It looks like AD500-9 have higher breakdown voltage, than AD500-8. Also, you need to remade PCB for any leaded APD.
I think you need to find out why your circuit can't produce higher voltage.

br76530067, 2021-08-18:
Oh, thank you for your reminder, after several times of searching in Taobao, I found one AD500-8.

"I think you need to find out why your circuit can't produce higher voltage."
I will try more.

closed 2021-11-01 by br76530067