iliasam/OpenTOFLidar #11: After installing the AD500-8

br76530067, 2021-08-20:
Hi, after I install the AD500-8 and lens. I got this picture.
[image]

I do not know how to set the lens at a proper focus point. I will try to find some answers in your wiki.

iliasam, 2021-08-21:
I can see that you had set "BIN Length" to 7.0 mm. It is not looking OK -  TDC BIN is 90 ps, which give theoretical value 13.6 mm.
Calibration curve should contain at least 10 points with different signal amplitude but with the same distance.

I have published some information about lens focusing: https://github.com/iliasam/OpenTOFLidar/wiki/Lens-focusing
See also: https://github.com/iliasam/OpenTOFLidar/wiki/No-optics-tests

br76530067, 2021-08-28:
I have been busy recently, and now start working on it

br76530067, 2021-09-06:
Hi, After bought a CCD industrial camera, I got the laser spot at 1.45m as below picture:
[image]
and could see the spot have about 12mm long, and I am buying an extender of lense to try more.

iliasam, 2021-09-06:
Spot is looking OK.

br76530067, 2021-09-11:
Hi, Am now is working on how to transplant to my MEMS mirror to detect distance.

However, I could only detect 1-2 Meter span laser

My structure is like below:
[image]

and could see the Spot is too large to focus on the center of the MEMS mirror:

[image]

[image]

The rotate mirror is the area in the red circle. And the rotational angle is about 15 ° ~ +15°, so I may need a receiving lens to get a larger receiving field.

[image]

I guess there are three main reasons causing why I can't get further distance when detecting:

1, the Spot is too large. (I am buying a M12 laser focuser to try)
2, the receiving Lense can not get the more large receiving field.
3, the APD500-8 is for 805nm, however, our Laser is 905nm, the strength of receiving diminished.

br76530067, 2021-09-11:
Could you help me to give some advice? because I am not familiar with the optics...
[image]

br76530067, 2021-09-11:
[image]

iliasam, 2021-09-11:
Hello.
Have you tested lidar with the same optics as I recommended in BOM (without MEMS)?
I think **you need to achieve good results without MEMS**, and try to install MEMS only after that.
All initial optics alignment should be done without MEMS too. 

"3, the APD500-8 is for 805nm, however, our Laser is 905nm, the strength of receiving diminished." I had never had problems with APD500-8. Lidar range was at least ~15m with APD500-8.

I'm sorry, I'm have never worked with MEMS mirrors. I have thoughts about using MEMS mirror in lidar, but I either don't know right way to use high power IR laser with MEMS . 
High power IR laser usually have very long enough emitting zone (comparing to low power laser diodes). Long emitting zone means that image is not symmetric, so it is hard to focus, and such laser have too big divergence angle.

If you take lens with small diameter (same with MEMS diameter), this lens would have small focus value, so divergence angle would be be very big, so angular resolution would be low.
If you take lens with big diameter (much bigger that MEMS diameter),  this lens could have big focus value, so divergence angle would be low, but as you can see - you will lose some emitted power. Maybe power loss would be not really dramatic - it need to be calculated accurate.
 
 To get better results, you probably will need really complicated lens and lens holders - like custom cylindrical lenses -  to change shape of lasers beam. Maybe you will need some optics after MEMS mirror.
 
 Another way is to use laser diode with less power, like MIR-905-25: https://www.alibaba.com/product-detail/TO18-Pulsed-Laser-Diode-905nm-25w_1600170943009.html 
 According to specs, it have "75*10 um" emitting zone (SPL LL90_3 have it 200x10um), so I suppose that it would have smaller spot.

 "2, the receiving Lense can not get the more large receiving field."
 As I know, all lidars with MEMS have very short focus lenses (like fish-eye) (so moving spot would be very small and it would not leave APD).
 But fish-eye lenses usually have too small aperture, so it could be hard to find suitable for Lidar.
 
  P.S. I'm not really sure in information above.

br76530067, 2021-09-11:
[image]
This is what I got without the MEMS. The actual distance should be 10M.

And I will buy a lower power laser diode to try.

Maybe I should hold an optics platform to test a variety of lenses.

iliasam, 2021-09-11:
"TOF std. dev" is looking too big.
You can see here my measurements: https://github.com/iliasam/OpenTOFLidar/wiki/Resolution-of-the-LIDAR
My value was near 1-2 cm at 10m comparing to yours 70 cm.
It could could be useful to no optics tests: https://github.com/iliasam/OpenTOFLidar/wiki/No-optics-tests
TIA is really sensitive to static electricity, when it got broken, lidar sensitivity drops significantly (20-100 times).

br76530067, 2021-09-11:
[image]
[image]

When it was at about 0.8M & 1.5M could get the above picture.

Because my 10M test was not on the formal testing object (actually a wall), and I will follow your advice to test if the TIA is broken.

closed 2021-11-01 by br76530067

woaigithuber, 2022-02-09:
In the process of making a PCB board, we have doubts about some component parameters in the file TOF_rangefinder_v4b.xlsx, such as the size and package format of L3, L5, P15, C37, X1, X2, etc. If it is convenient, can you provide us with some detailed instructions about the components when making the PCB, we would be very grateful for your help.

iliasam, 2022-02-13:
You can see from BOM, that:
L3 - QM21FN100M80L  - 0805[2012 metric] 
L5 - LQH31CN470K03L - 1206[3216 metric]
You can check them at mouser.com
P15 - is PLS-3
C37 - any polarized capacitor 100uF with 5mm between pins, >10V
X1 - any leaded cylindrical 32.768kHz quartz for with 3mm between pins, like this one: https://aliexpress.com/item/1005003048673233.html
X2 - package SMD06035C4 - 6x3.5mm, SMD05032C4 package is suitable too. 8MHz. 4 pins.