iliasam/OpenTOFLidar #12: Low-cost 3D scanner

DeRuiY, 2021-12-20:
I have been able to get a two-dimensional point cloud. I would like to DIY a low-cost 3D scanner based on your design. What do you recommend? thanks  
[image]

iliasam, 2021-12-20:
Firstly I need to say that you have certain problems with calibration. Probably you need to set correct "BIN Length" value.  

If you are planning to make 3D scanner, you need to add another drive to rotate whole LIDAR.   
Example: https://youtu.be/3SnqrfDCc70  
P.S. I'm interested in any information about this Lidar from others, so your experience with this Lidar will be really interesting for me.

iliasam, 2021-12-30:
@DeRuiY  Please, close this issue if you don't have other questions.

closed 2022-01-26 by iliasam

DeRuiY, 2022-02-15:
Hello, I have come, I have been thinking about the ranging problem of TDC-GP21, because I used TDC-GP22 for a while before. Why do you want to measure distance this way, instead of using
[image]
instead of the way below
[image]
[image] .Thanks

iliasam, 2022-02-15:
Hello. 
Sorry, but I don't understand your question.
What is the "ranging problem of TDC-GP21" ?
My correction is used for compensating pulse amplitude/changing (exponential part) and for zero offsets compensation.
See https://github.com/iliasam/OpenTOFLidar/wiki/TDC-data-processing