iliasam/OpenTOFLidar #13: Sensor Readings

motius-ggaryuk, 2022-01-26:
Hi, 
amazing project!
I am wondering what are the raw values you get from the photo receiver. 
In particular, can you see multiple responses with different intensities for a single laser beam?
For example when looking at a glass surface can you see the response of the glass and the response from a structure behind the glass?

iliasam, 2022-01-26:
Hello.
Current firmware do not support multiply stop events. 
I'm not sure that current TDC schematic support it too.
Laser pulse length is big enough (>20 ns), also TIA can work in overdrive mode, so resulting pulse length get increased.
20ns is corresponding to 3m, so multiply stop events support with current hardware seems useless for me.

motius-ggaryuk, 2022-01-28:
Thank you very much!

closed 2022-01-28 by motius-ggaryuk