iliasam/OpenTOFLidar #14: Question on Lidar

YileAllenChen1, 2022-03-03:
This is an amazing project, you did a great job! I have a few questions regarding some of the details.

In this [image](https://i.ibb.co/W6KjFtM/IMG-7432a.jpg) I see you are connecting an antenna to the orange pi PC, could you please tell me what this is for? You mentioned that the ROS node and hector SLAM runs on the orange pi, but are you still transmitting any data from the orange pi to your PC?

Also if you're running the SLAM software on the orange pi, how are you displaying the scanning results in real time? Since the orange pi is not connected to a monitor.

Thank you in advance for your time and help!

iliasam, 2022-03-03:
Antenna is connected to WIFI dongle, which is connected to  Orange pi PC.
Orange pi PC is connected to local network via this WIFI.
My x86 PC with "rviz" running is also connected to this network. So all SLAM is running at the Orange pi PC but I can view real-time results at my  x86 PC.
Hardware configuration is the same as here: https://youtu.be/RW57MEIpBHI?t=11  but in this TOF Lidar project ROS Master is Orange pi PC.  
  
See also: http://wiki.ros.org/ROS/Tutorials/MultipleMachines

closed 2022-03-09 by iliasam

YileAllenChen1, 2022-03-10:
Thank you for the help!