iliasam/OpenTOFLidar #16: I want get mavlink.dll source code

WangyunhuiSeasun, 2022-04-13:
I use mavlink_tof.xml to generate mavlink.dll , but it is different from your mavlink.dll。
Or where should I go to get the project to your mavlink.dll。
thank you。

iliasam, 2022-04-17:
This XML was used for generating mavlink.dll: https://github.com/iliasam/OpenTOFLidar/blob/develop/Firmware/MavlinkXML/mavlink_tof.xml
I have checked mavlink.dll files in this repository, they are actual.
I can't give you sources of mavlink.dll because they are generated dynamically by mavlink builder.  
As I can see, this DLLs contains some variable information (probably build time), so they will differ in a small part, even if they are generated from the same mavlink_tof.xml.

iliasam, 2022-05-09:
Any questions?

closed 2022-07-06 by iliasam