iliasam/OpenTOFLidar #17: measurement concern of the system

ClintSun-Git, 2022-07-06:
hello, you have done a great job on this project! I'm working on a sanning lidar based on polygon mirror, I have a critical qustion. Did you ever meet an issue that the when laser rays emits to the mirror, some rays is then directly deflected to the APD lens and received by APD? In that case the distance measured is not right due to this kind of light interference. In my design, I have to use a plastic isolator to block this interference. I don't know is there such interference in your design, or the lidar system is not effected by this kind of interference due to mirror flatness, TIA threshold, or TDC principle, etc. 
Thank you and good luck!

Kind regards,
Clint

ClintSun-Git, 2022-07-07:
hello,
I upload a structure picture,
[image]
You can also watch this teardown video of valeo ibeo used on audi, the structure is very similiar.
https://www.bilibili.com/video/BV1dL411A77G?spm_id_from=333.999.0.0&vd_source=899f9c339bd72b974a002ec80689d231. 
Did you ever make optical simulation with zemax or tracepro? It's important but very difficult.
By the way, I see MAX3658 is not good enough for your design, maybe you can try MAX3665.

iliasam, 2022-07-07:
"did you ever think about any commercial possiblity of your design or product?"
I think that similar or best Lidars are already exists at the market. LD06 is an example.
Yes, I'm living in Russia now. I don't know about commercial Russian Lidars. I know that non-scanning laser rangefinders are produced here, I heard that a custom-made special Lidars are produced for special industrial devices.
I think that this is because it is to hard to compete with USA/Europe/China Lidars here.

ClintSun-Git, 2022-07-08:
I think the advantage of LD06 is small size and cheap price, it's a very competitive product, but the technical design is poor, the parameters and performance are not clear enough or as good as they say. 
I know very less about Russia than USA/Europe, but I know Europe and the world need oil and gas from Russia, and Russia is powerful because you have very strong weapon, and everything is in a dramatic change now. I also think your hard work for open lidar is worthy of respect.

ClintSun-Git, 2022-07-19:
hello, I'm here again and I met an issue in evaluation. I measured START and STOP signals with oscilloscope, in close distance(<30cm), when i moved the target object, the gap between these two pulses was not changed. I use hamamatsu APD and MAX3665. What's the possible reason of this issue, does it possibly caused by TIA? Thank you
[image]
[image]
!

iliasam, 2022-07-19:
Hello.
"when i moved the target object, the gap between these two pulses was not changed"
What is the distance of movement?
You can write me directly to the iliasam@mail.ru

closed 2022-07-21 by iliasam