iliasam/OpenTOFLidar #18: Questions on encoder 安定

buzzzzm, 2023-07-04:


buzzzzm, 2023-07-04:
hi,iliasam,
I have tried to follow the Electronics-assembling notes and test my advice in no-scanning mode.
I found my encoder can’t detect the disc with 65 silts, so I replaced it with a 20 silts disc.
[image]
[image]

When I use the hex file you give, the “testing utility” showed “0x8”-wrong number of encoder events. 
[image]

So, I changed the parameter “ENCODER_HOLES_CNT” to 20.
But the data is not displayed when I run the “testing utility” again. why???
[image]
[image]

Can I change the disc to a 20-slits one?  
Or what model of the encoder can detect a 65 silts disc? I want to know the product model. The one I bought doesn’t work well.

iliasam, 2023-07-04:
First question - is your Lidar working well in non-scanning mode?
Possibly FW should work with encoder disc with 20 slits, if you **rebuild** the firmware.
Have you checked that your encoder is working well with an oscilloscope?
I can't give you model of my encoder sensor, because I take it from old printer. 
Maybe RPI-352 sensor will work well.

buzzzzm, 2023-07-06:
First, I tested in no-scanning mode, is it working well?
The pictures are calibration page and no optics test results.
[image]
[image]
[image]
[image]
[image]

I checked the encoder with an oscilloscope.

[image]

I will buy the RPI-353 sensor and test with 65 silts disc.

iliasam, 2023-07-06:
Your results in non-scanning mode are looking OK.
Oscilloscope picture from the encoder is looking well, too. I can't say now what is the problem.
You can check MCU pin 10 with oscilloscope to ensure that encoder pulses are getting to the MCU.

iliasam, 2023-07-06:
[scanning_fw_20_holes.zip](https://github.com/iliasam/OpenTOFLidar/files/11966272/scanning_fw_20_holes.zip)
You can try this FW - I build it for encoder with 20 holes. Also I had increased ENCODER_TIMEOUT_MS value to 25 ms.

buzzzzm, 2023-07-06:
I downloaded the hex file you just sent, but the testing utility doesn't work.
And the BLDC motor rotates at a faster and unstable speed. 
[image]
And I downloaded the hex file "scanning_fw_pcb_v4" again, the state mask is 0x8.
The motor starts fast and then slows down, eventually stabilizing its rotation.
[image]

[image]

iliasam, 2023-07-06:
You can try to use "scanning_fw_20_holes" FW
and set speed of the motor manually at the "Motor Settings" page - see "Motor Manual PWM" -> "Set" button.
Do "Motor Speed" value is measured OK in this mode?

iliasam, 2023-08-02:
Do you have any progress with your problem?