iliasam/OpenTOFLidar #19: I have a question about LidarTestingUtility.

tmdwhsh, 2023-07-18:
Hi, iliasam.
I have uploaded 'not_scanning_fw_pcb_v4.hex' and the
and noticed that the LED is blinking.
I also checked the 'Laser_Pulse' of the TDC and found that a pulse of about 3V is generated.
I also checked that the voltage of the Vlaser is changing the same as what I set in the Testing Utility.
However, if you look at the picture I attached, you can see that many of the items in the Rangefinder State have been changed to
indicating a 'N/A' state.
Since the voltage is set normally and various values are recorded through 'Start', it seems that the serial communication is working well, but can you tell me why 'N/A' appears?
[image]

iliasam, 2023-07-18:
Hello. Can you tell me what is the release of firmware that you are using?

tmdwhsh, 2023-07-19:
I used pcb v4 and uploaded 'release2 - not_scanning_fw_pcb_v4.hex'

iliasam, 2023-07-19:
Testinng Utility is decigned for beeing used with firmware from "Release 2" so it is strange that you are getting state data from "Release 1".
You can try to use this dll with "Testinng Utility".
[mavlink.zip](https://github.com/iliasam/OpenTOFLidar/files/12092707/mavlink.zip)

tmdwhsh, 2023-07-19:
I figured out my problem.
When I uploaded Release1, the laser works and N/A appears, but the
When I uploaded Release2, the laser doesn't work, but the N/A is gone.

Any idea what I can do to fix the above (for the laser to work in Release2 as it did in Release1)?

I'll try to test with your DLL file and Release1 first. 
Thanks a lot!!!

iliasam, 2023-07-19:
I can see that "Bad communication with TDC after init" bit is set.
I recommend you to check SPI TDC lines soldering at MCU and TDC side.

tmdwhsh, 2023-07-20:
I think it was a soldering issue.
We rechecked the SPI line and found that it was communicating with the laser in Release2.
Thanks!

https://github.com/iliasam/OpenTOFLidar/assets/43981588/7d3b56c7-c1aa-471f-9211-44c28e3576df