iliasam/OpenTOFLidar #20: I have a problem with Vlaser waveforms

tmdwhsh, 2023-07-25:
[image]
[image]
[image]
I measured the VLaser at the J2 terminal, but could see that it did not match the ideal waveform you posted (LaserV_10~16V).
I could see that the amplitude was very large, around 13V, and the overall shape didn't match either.

I measured from J2 using the U.FL terminal and the measurement instrument was a Tektronix TDS 2012C.
[image]

iliasam, 2023-07-25:
How do you connect your U.FL cable to the oscilloscope?
 Is D7 installed right?

tmdwhsh, 2023-07-25:
I didn't have a U.FL to BNC cable, so I connected a U.FL to SMA cable and an SMA to BNC cable to connect the U.FL terminal of J2 to the oscilloscope.
[image]

[image]

The photo shows the D7 and everything else installed.

Could the above problem occur if the D7 was not installed properly?

I could try replacing the D7 with a new part or reversing the orientation of the D7 and installing it.
[image]

iliasam, 2023-07-25:
"Could the above problem occur if the D7 was not installed properly?"
Maybe.
Can you give me a photo of your laser diode (PCB from other side)?

tmdwhsh, 2023-07-26:
[image]

[image]

I was able to see the laser light up using a camera and IR detector card.

I will try to replace the D7, starting with what you said

iliasam, 2023-07-26:
"I reversed the connection of the D7 and noticed that the amplitude of the waveform decreased, but it still doesn't look normal."
You need to install diode as it is drawn at the schematic, this diode has a marking line near cathode.

Did you measured with voltmeter voltage at the R18, C17?
Have you selected right divider (1:1) at the oscilloscope channel which is connected to the UFL?
Is Q2 transistor soldered well?
You can check with an oscilloscope probe voltage shape at all lines of Q2.

tmdwhsh, 2023-07-26:
[image]
[image]
First, I thought that the flat part on one side was the Cathode, just like a normal LED, so I installed D7 accordingly, but after looking at the data sheet of SPL90_3, I realized that the flat part was the Anode, so I installed D7 accordingly.
But even after that, the waveform was still the same.

First, I checked the voltage of R18 and C17. 
R18 was measuring a very low voltage,
[image]
C17 was measuring a high voltage of 18-20V.
[image]
[image]

Next, I measured the voltage across the Drain, Gate, and Source of Q2.
For the Drain, we found the same voltage as C17,
For the Gate, it measured around 20V as shown below.
[image]
For the Source, it was measured as shown below.
[image]
The soldering condition of Q2 looks normal.

tmdwhsh, 2023-07-27:
[image]
First I set the Vlaser to 16V, got a clear oscilloscope screen, and measured the Drain, Gate, and Source voltages again. 
[image]
Now I'm going to remove Q2 and measure it the same way and upload it again.

tmdwhsh, 2023-07-27:
This is the waveform of pin 2 of DA6 after removing Q2.
[image]

iliasam, 2023-07-27:
"First I set the Vlaser to 16V"
But did you measure Vlaser with voltmeter? MCU do not measuring Vlaser, so you need to check that electronics is working OK by yourself.
I can see that you did something, so the shape of voltage at the gate changed (photo from 14:46), now it is looking OK.
Other photos now are looking similar to mine. So you can install Q2 back.
_DA6 pin 6 is not a Gate line._
Did you try to remove UFL cable and measure voltage at the J2 directly with oscilloscope probe? I recommend to use "ground spring" for that: https://electronics.stackexchange.com/questions/40420/what-is-the-name-of-this-springy-type-oscilloscope-probe-accessory

tmdwhsh, 2023-07-27:
Sorry,, That was DA6 pin 2.

First, I verified with a multitester that the Vlaser is 16V.
[image]

Next, I measured with the default probe(X10) instead of the U.FL.
[image]
[image]

iliasam, 2023-07-27:
I still don't understand what is shown at the photo from 16:44.
Looks like that yellow CH1 is still a signal from UFL, but UFL cable must be disconnected from PCB for tis measurement.

tmdwhsh, 2023-07-27:
I assumed the small terminal next to J2 was the same as the UFL terminal, so I measured it as shown below.
[image]
[image]

iliasam, 2023-07-27:
UFL cable must be disconnected because it create affect to a signal - cable has its own capacity.
It is OK to measure signal at the pin of the UFL connector as you did.
And I repeat again - use "ground spring" to get better results. "Ground spring" is short, so it has less inductance.

tmdwhsh, 2023-07-28:
I took another measurement using a ground spring! (I ordered one from Aliexpress today, but it took a while to ship, so I made a similar one using copper wire)
Does this represent a good enough waveform to proceed to the next step?
[image]
[image]
[image]