iliasam/OpenTOFLidar #21: What should I do when a bit2 error occurs?

tmdwhsh, 2023-09-11:
[image]

I got the error "bit2 (0x4) - Wrong hits number from laser comparator" as shown in the photo above and I can't proceed any further.

What can I do to fix a situation like this?

iliasam, 2023-09-11:
Try to check DA5 output with oscilloscope.

tmdwhsh, 2023-09-13:
The above problem occurred while creating the second Tof Lidar!
I'll take your feedback into consideration and keep working on it.

I am in the process of calibrating my first Tof Lidar, measuring distances and calibrating it. 
[image]
I used the Non-Scanning Firmware to set the Refence Distance and compared the value measured by the Testing Utility with the actual measurement using a tape measure and found that I got a very accurate value.

However, due to an incorrect calibration method, the Calibration Graph does not come out ideal (similar to yours) and the Scanning Test Utility does not come out correctly. 

Firstly, can you give me a step by step on how to calibrate?
We recorded the data in the following way
1. Upload Non-scanning Firmware.
2. obtained the actual reference length for measurement using a tape measure.
3. set a target for measurement using a wide plate.
4. The measurer stands with the plate at a close point (Ex: 1M), and the Lidar is pointed at the point, and the Reference Distance is set to 1000mm.
5. Start Slow Decoding, and Pressed Start button(Capture Button, top left). 
6. The measurer moved away sequentially (0.5M increments), changed the Reference Distance to match, and pressed the Start button (Capture Button) again.
7. repeat the process of Step 6 and use it to get the Calibration Graph and Coefficient.
[image]
8. After uploading the scanning firmware, send the coefficient obtained in Step 7 to the MCU.
9. Scanning was performed using the Scanning test Utility.

I'm afraid I've done it the wrong way.

iliasam, 2023-09-13:
Please do not mix different questions in one issue!

"Firstly, can you give me a step by step on how to calibrate?"
Please, read Wiki carefully: https://github.com/iliasam/OpenTOFLidar/wiki/Walk-error-calibration
Yes, you do calibration wrong.

tmdwhsh, 2023-09-15:
[image]
[image]
[image]
I tried calibrating again and got the same results as above.

iliasam, 2023-09-15:
Did you install "Reference Plate" to your Lidar?
This plate is not described well in Wiki, but it was mentioned in the original article: https://habr-com.translate.goog/ru/articles/485574/?_x_tr_sl=auto&_x_tr_tl=en&_x_tr_hl=en&_x_tr_hist=true
Distance between plate and rotation axis is set in code: https://github.com/iliasam/OpenTOFLidar/blob/f3f400fe56a28f5006bd249ed1249cb4261fe623/Firmware/scanning_fw/config.h#L24

tmdwhsh, 2023-09-30:
I checked the Reference Plate as you said, and after mounting the Reference Plate and adjusting the zero point of the Encoder Disk appropriately, I confirmed that the scanning result is observed normally.
https://github.com/iliasam/OpenTOFLidar/assets/43981588/1801a132-f785-4406-82c5-934e3e41a8c1

closed 2023-09-30 by iliasam