iliasam/OpenTOFLidar #22: Laser constantly pulsing in no scanning mode?

taliaxu09, 2023-11-07:
Hello! Great project! Thank you so much for developing this.

I originally thought that the laser is only pulsing when "start" or request batch is pressed in the PC utility in no scanning mode, however, using a photodiode, it seems that the laser is constantly pulsing at 1KHz even with no command from PC utility. I just wanted to check that this is as designed?

iliasam, 2023-11-07:
Hello.
You are right, but it is mentioned here: https://github.com/iliasam/OpenTOFLidar/tree/develop/Firmware in readme.md
Blinking is controlled from main.c

taliaxu09, 2023-11-09:
That makes sense! I was a bit confused by the utility because I thought maybe pressing start initiates the 1 kHz pulse until it's done sampling

I measured the waveform at J2 and I think this looks pretty good?
[image]

Thank you again for making this project!

iliasam, 2023-11-09:
I think that shape of the signal is OK.

closed 2023-11-20 by iliasam