iliasam/OpenTOFLidar #23: I have a question about LidarTestingUtility.

pikasm, 2023-11-20:
Hello, there was no pulse in my previous TDC. At that time, the request of the program was correct. Now TDC has a pulse and it becomes N/A. Does this mean that my single chip microcomputer is broken?
[image]

iliasam, 2023-11-20:
Hello.
Is MCU flashed with right firmware?
Check your UART connections.
Check that MCU quartz oscillator is soldered well, wires from  quartz  to MCU is OK.
Do LED blinking? How fast?

pikasm, 2023-11-20:
I tested the frequency of LED. It's 5HZ.

pikasm, 2023-11-20:
[image]

pikasm, 2023-11-20:
Here are my test results.
[image]

pikasm, 2023-11-20:
Should I check if there is something wrong with MCU next? 
But my TDC pulse is correct, DA6 also has pulses, J2 also has pulses. 
Should I reinstall a DD2?

pikasm, 2023-11-20:
Then I'll check it. Thank you.

pikasm, 2023-11-21:
I replaced the single chip microcomputer and got a signal. Thank you so much. I am so happy.

closed 2023-11-21 by iliasam