iliasam/OpenTOFLidar #25: Increasing pulse width and frequency

taliaxu09, 2024-01-10:
Hello, I would like to increase the power output by increasing the pulse width to be as close to the maximum (I believe 100 ns?) as possible and also increase the frequency from 1 kHz to maybe 5 kHz or 10 kHz.

Would you point me to where I could change this in the code? Can I increase the pulse width by using a larger capacitor?

iliasam, 2024-01-10:
Hello.
1kHz is just test signal frequency, see main.c of "not_scanning_fw" project (TIMER_ELAPSED(timer_1ms) part).
"scanning_fw" reach 10kHz pulse frequency at 15 RPS motor speed.

You can try to increase pulse width by using a larger capacitor, but be careful!

taliaxu09, 2024-01-10:
Thank you! What would be the best software version / setup to compile the code?

iliasam, 2024-01-11:
I don't know what is your hardware, do you need scanning or not, so I can't help you.

taliaxu09, 2024-01-11:
I built it as yours, and I don't need scanning and the motor, I am just using the laser.

I just wanted to know the best setup to be able to compile the code (I only used Keil uvison before) because I saw in another thread that there might be problems if the version is different.

iliasam, 2024-01-11:
I use IAR 7.50 for building this FW.

iliasam, 2024-01-30:
Software counters, like "timer_1ms", are based on "ms_tick" variable, that is incremented in SysTick_Handler(), which is called by system timer with 1ms period.
This timer is configured by SysTick_Config() function in "hardware.c".
A lot of code here is expecting that this timer has 1ms period.

One option is use hardware CAPTURE_TIMER, that is used in "scanning_fw". It can generate interrupts with needed frequency.

Another way - you can use hardware_dwt_get() function, that return MCU tick count (72 ticks per 1us). 
You can see dwt_delay_us() function.
But this kind of delay will lock program execution.

closed 2024-01-30 by taliaxu09

ShanLan12, 2024-06-29:
Hello, I have a question for you, I recently came across a piece of code in my project which uses the MS1005 chip using the stop1 channel and stop2 channel connected to the same pin. A rising edge trigger and a falling edge trigger can be used to get the pulse width and use this pulse width to calibrate the distance measured. May I ask why it does this and how it calibrates the distance
[image]
[image]
Thank you!

iliasam, 2024-06-29:
[ShanLan12](https://github.com/ShanLan12) Please, create new issue for your question.