iliasam/OpenTOFLidar #26: Problems about motor rotation and rotation program

pikasm, 2024-01-12:
Hello, I installed the motor and downloaded the scanning program today, but my motor doesn't rotate after downloading. After I set the PWM value of the motor, the motor will rotate a little bit. I want to know what the reason is. Can you help me?

https://github.com/iliasam/OpenTOFLidar/assets/147833145/4aeb1ff4-fe31-4ae5-9ec5-f9ace65a8822

https://github.com/iliasam/OpenTOFLidar/assets/147833145/b459526b-1cde-44e1-82a1-86eb3a4d69fa

pikasm, 2024-01-12:
[image]
Disconnecting the motor can change the PWM. Does this mean that the system does not provide suitable current for the motor?

pikasm, 2024-01-12:
[image]

pikasm, 2024-01-12:
But my power supply is an instrument, and it doesn't indicate that the current is insufficient. Is it possible that the code wheel of my encoder is not the corresponding number of holes, and the motor will continue to run after I need to change it?

pikasm, 2024-01-12:
[image]
[image]

pikasm, 2024-01-12:
I changed the number of holes, but the motor still doesn't work.

iliasam, 2024-01-12:
When you set PWM manually, motor is not controlled by encoder, it must work in any case.

As I said - check the power line with an oscilloscope when motor is connected.
Motor can take >0.7A at startup.
I had used such board for motor testing in my experiments: https://aliexpress.com/item/1005006219691535.html

pikasm, 2024-01-22:
I tested the motor, and the initial current was really not big enough. I changed the power supply and the motor can rotate normally. Next, I should install the reference board and set the encoder zero angle. I want to know where the zero angle of the encoder should be stuck. Is it the slit irradiated by the encoder when the lens vertically irradiates the reference plate?

pikasm, 2024-01-22:
I changed my encoder, and the number of its slits is 100. What values do I need to change in the rotation test program?
[image]

iliasam, 2024-01-22:
See this page: https://github.com/iliasam/OpenTOFLidar/wiki/Encoder

pikasm, 2024-01-23:
I installed a reference board and a zero-angle reference mark. I opened the program like this. What else do I need to do next?
[image]

iliasam, 2024-01-23:
Check encoder output with an oscilloscope.
Ensure that zero pulse is visible.
Ensure that signal amplitude is > 1.2.

pikasm, 2024-01-23:
I checked these two values, and there is no problem. Maybe I should change the start angle and stop angle?Because my code dial is 100 yards.

iliasam, 2024-01-23:
Can you publish your oscilloscope screen image with zero pulse visible?
I don't think that problem is in software.
State mask 0x8 is problem with encoder pulses.

pikasm, 2024-01-23:
[image]
[image]
This is my encoder image, and you can see that the corresponding zero angle is empty.

iliasam, 2024-01-23:
You can see here: https://github.com/iliasam/OpenTOFLidar/wiki/Encoder
that "zero pulse" is keeping high polarity with my hardware. 
Zero position -> no light -> no current through opto-transtor -> Encoder output is pulled high by pull-up resistor.

pikasm, 2024-01-24:
I replaced a new encoder, and it showed radar images, but all of them were wrong points. I wonder why?
[image]
[image]
[image]

iliasam, 2024-01-24:
State mask: 0x28
You can see here: https://github.com/iliasam/OpenTOFLidar/wiki/State-Mask-bits-description
bit3 (0x8) - Wrong number of encoder events. Encoder is dirty or wrong installed
bit5 (0x20) - Mirror speed is too low or too high
There is problem with your encoder.

pikasm, 2024-01-24:
Should I change the encoder next? Does my code wheel need to be replaced?

iliasam, 2024-01-24:
Check output of the encoder with oscilloscope (at the line at main Lidar PCB).
Ensure that opto-interrupter output in at high state when opto-interrupter light is closed.
You can try to use logic analyzer to capture data from encoder and calculate count of pulses between two zero events.

pikasm, 2024-01-25:
I observed the value of the oscilloscope, perhaps because the diameter of the code wheel is too large and the waveform is a little distorted. Should I replace the code wheel?
[image]

iliasam, 2024-01-25:
Maybe yes. You can try to change position of the opto-interrupter sensor.

pikasm, 2024-01-29:
I replaced the encoder and encoder again, and now the display is like this. I want to know why there is no image.
[image]

[image]

iliasam, 2024-01-29:
Have you changed "ENCODER_HOLES_CNT" value in your code to "20"?

pikasm, 2024-01-29:
yes

pikasm, 2024-01-29:
[image]

pikasm, 2024-01-29:
Do I need to calibrate the walking error again after downloading the scanning program?

iliasam, 2024-01-29:
Try to increase ENCODER_TIMEOUT_MS value to 10.
"Do I need to calibrate the walking error again after downloading the scanning program?"
I think - no.

Please make a screenshot with LidarScanningTest with "Pointer Angle" value near 180 deg.
[image]

pikasm, 2024-01-29:
[image]
I don't know why, it has no angle value.

pikasm, 2024-01-29:
I have increased the encoder timeout millisecond value to 10.

iliasam, 2024-01-29:
Strange situation.
It looks like that you are receiving "DIST_MEAS_NO_SIGNAL_VALUE". This happens when you have TDC timeout.
Did you try to run "not_scanning_fw" with mirror installed?

pikasm, 2024-01-30:
I downloaded the not scanning program again, and the distance measurement is effective, but the error is 1cm-2cm larger than before.

pikasm, 2024-01-30:
My measurement under the installation mirror is like this

pikasm, 2024-01-30:
[image]

pikasm, 2024-01-30:
[image]

pikasm, 2024-01-30:
Is it possible that I am using the wrong iar version? I used 9.30.1.

iliasam, 2024-01-30:
Measurements with mirror installed are looking OK, so it looks like that distance measurement HW is working OK.
My IAR is 7.50, but I don't think that problem is connected with that.
Can you load "scanning_fw" again and check with oscilloscope that you have pulses at "LASER_PULSE" line (motor must be running, encoder must be connected)?

pikasm, 2024-01-30:
I found the problem. There was no laser pulse. After I downloaded the not scanning program, there was a laser pulse. After downloading the not scanning program (after downloading the not scanning program), there was a flashing pulse signal at first, but when I turned on the radar rotary scanning program, there was no correct point, and the pulse disappeared at this time. How should I solve this problem?

pikasm, 2024-01-30:
I found the problem. There was no laser pulse. After I downloaded the not scanning program, there was a laser pulse. After downloading the scanning program (after downloading the not scanning program), there was a flashing pulse signal at first, but when I turned on the radar rotary scanning program, there was no correct point, and the pulse disappeared at this time. How should I solve this problem?

iliasam, 2024-01-30:
Sorry, I don't understand your description well.
As I can understand, you say, that "LidarScanningTest" is making pulses disappear?
Can you describe step by step, what you are doing?

Does motor start running after power on without any commands from PC with "scanning_fw" ?
Is the problem repeatable after Lidar power reset?
Do problem exists if you disconnect wire from PCB RX connector (so MCU can only transmit, but not receive)?

pikasm, 2024-01-30:
I mean, after downloading the scanning program, the pulse signal of the laser began to disappear gradually. Causing me to have no value after rotation measurement.

pikasm, 2024-01-30:
Every time I use the motor, I will set a pwm value and then measure the rotation; The motor does not rotate when it is energized (it stops every time it is energized).

pikasm, 2024-01-30:
There is really no pulse, and I don't know why.

iliasam, 2024-01-30:
"Every time I use the motor, I will set a pwm value and then measure the rotation; The motor does not rotate when it is energized (it stops every time it is energized)."
I don't understand what do you mean by that.

pikasm, 2024-01-30:
I mean, every time I connect the power supply, the motor will not rotate, and then I will use the LidarTestingUtility program to set the PWM value of the motor, and then the motor will start to rotate.

iliasam, 2024-01-30:
You can press yellow "Save to Flash" button at the "Main" page of the "LidarTestingUtility" to save motor settings, so motor will be enabled after Lidar power on.

pikasm, 2024-01-30:
[image]

[image]

pikasm, 2024-01-30:
I have saved the pwm value of the motor, and there is really no laser pulse when using it.

iliasam, 2024-01-30:
Can you check PA2 pin with oscilloscope (P8 pad, see ["PCB_3D_view_up.png"](https://github.com/iliasam/OpenTOFLidar/blob/develop/PCB/PCB_project_v4/PCB_3D_view_up.png)). Motor must be running.
This is test output, MCU set this pin to 1 during reading measurement from TDC.

pikasm, 2024-01-30:
[image]
I measured its value, and there is no problem.

iliasam, 2024-01-30:
This is looking like a mess, can you give a single measurement, with 1ms in a horizontal grid step?

pikasm, 2024-01-30:
[image]
[image]

iliasam, 2024-01-30:
I need "1 ms", not "10 us" horizontal grid step.

pikasm, 2024-01-30:
Sorry, I misunderstood. Please wait for me for a while. I'm outside now. I'm very sorry.

pikasm, 2024-01-30:
[image]

pikasm, 2024-01-30:
Do you have any ideas?

pikasm, 2024-01-30:
[image]

pikasm, 2024-01-30:
[image]

iliasam, 2024-01-30:
P8 is looking OK. Period is really need to be near 110us if mirror has 12Hz frequency.
So MCU is processing encoder data OK, calculate motor speed OK, one pulse is lasts 30us, so looks like MCU is really trying to read TDC values. So if you see long (>20us) pulses at P8, you must see synchronous laser pulses at  J2.
But I need more time for my own experiments to measure real duration.

You can see in "CAPTURE_TIMER_IRQ_HANDLER", how it is working.

As I advised, you can try to test Lidar with RX pad disconnected, so PC can't affect  to FW.

I build scanning FW for 20 holes, you can try it: [scanning_20_holes.zip](https://github.com/iliasam/OpenTOFLidar/files/14099306/scanning_20_holes.zip)

iliasam, 2024-01-30:
Always check "State mask" value in utilities. It is important thing to understand Lidar internal state.
"State mask" is updated by requests from PC only.

pikasm, 2024-01-30:
Thank you very much. Can you teach me how to use scanning FW for 20 holes?I'm a computer novice.

[92 more replies]

closed 2024-02-26 by iliasam